greiman/SdFat-beta #13: SD.open() is valid after removing SD card

gert3d, 2016-10-26:
In a datalogger sketch I use the line 
   myFile = SD.open( fileSD, FILE_WRITE ) 
to test whether a write can be done. After a couple of successful (checked) writes, I removed the SDcard and to my surprise the test: <    if (myxFile)   > remains true.
If I re-insert the SDcard no more data will be added, although the test remains to be true.
I am using this  SD Card module:  goo.gl/q0AeM2
Even the test < if (SD.exists(fileSD)) > remains true if the card is removed after successful write cycles
Same result for the SD.h library

gert3d, 2016-10-26:
It is probably due to hardware in Arduino UNO clone, since ProMini behaves differently: after removing the card an error is reported for opening FILE_WRITE.
However upon re-inserting the card the error remains. Resetting helps. Is that normal?

greiman, 2016-10-27:
Yes, you can't remove and insert the card.  Data is cached so an open may appear to work.

closed 2016-10-27 by greiman