greiman/SdFat-beta #15: errors with SdFatSdio in teensy 3.6

hespanha, 2017-03-24:
I'm getting intermittent errors with erase and writeBlock. To get the SdFormatter example to work, I did the following:
1) changed the define in line 16 to
#define USE_SDIO 1
2) changed line 42 to
SdioCard card;
(used to have SdioCardEX but I have not been able to make that work)

With these changes things go well, but almost always I get errors with the erase and writeBlock. To prevent this, I also changed:
3) around line 171 I changed the "if" to a "while"
    while (!card.writeBlock(bgn + i, cache.data)) {
      cout << 'e';
       //sdError("Clear FAT/DIR writeBlock failed");
    }     
4) around 423 I also changed the "if" to a "while"
    while (!card.erase(firstBlock, lastBlock)) {
      cout << 'E';
      //sdError("erase failed");
    }

The code eventually succeeds (after repeating the erases and write blocks a few time) but I cannot read the card. I'm also having trouble with other sketches with errors in erase and writeBlock.

greiman, 2017-03-24:
I can't help since it works for me.  Here is a high end Samsung 32 GB card with SdioCardEX.

> Type any character to start
> 
> This program can erase and/or format SD/SDHC cards.
> 
> Erase uses the card's fast flash erase command.
> Flash erase sets all data to 0X00 for most cards
> and 0XFF for a few vendor's cards.
> 
> Cards larger than 2 GB will be formatted FAT32 and
> smaller cards will be formatted FAT16.
> 
> Warning, all data on the card will be erased.
> Enter 'Y' to continue: Y
> 
> Options are:
> E - erase the card and skip formatting.
> F - erase and then format the card. (recommended)
> Q - quick format the card without erase.
> 
> Enter option: F
> Card Size: 32011 MB, (MB = 1,000,000 bytes)
> 
> Erasing
> ................................
> ................................
> ................................
> ................................
> ................................
> ................................
> ................................
> ...............
> All data set to 0xff
> Erase done
> 
> Formatting
> Blocks/Cluster: 64
> FAT32
> ............................................................
> Format done

It also works for SdioCard.

Have you tried other SD cards?

hespanha, 2017-03-24:
Thanks for the prompt reply. I will try a different card.

I’ll also try to figure out how come the teensy example works flawlessly, but I am having problems with this one. I assume that the usual way to write to the card does not use erase() but it should use some version of writeBlock() so it is strange that I get no errors with the teensy example. I’ll dig more into this and will let you know if I find any issue with the github code.

Best,
Joao

greiman, 2017-03-24:
Don't use code from the format example for regular applications to access raw blocks on the SD.

I only support API calls at the file system level.  I don't support card.writeBlock() as a standard API call.  

You should use file.write(buf, nbyte).

hespanha, 2017-03-27:
Thanks. I'm trying to do a data logger somewhat similar to the one in the examples, which uses erase and writeBlock to speed things up. 

Following your advice I moved back to file.write and the times are not bad, as long as I make sure that my writes are aligned with the 512 block boundaries (which makes sense). However, I'd still like to go faster.

greiman, 2017-03-28:
Erase does not speed writes with modern SD cards.  I should remove erase from examples.

Modern cards are designed for optimal performance with smart phones.  The cards assume massive contiguous writes using the 4-bit SDIO interface, not SPI.

SdFat can write at up to 19 MB/sec with a Teensy 3.6 board.

Here are Teensy 3.6 file read/write results as a function of transfer size.  This is the TeensySdioDemo example.

> size,write,read
> bytes,KB/sec,KB/sec
> 512,19291.70,20031.01
> 1024,19584.91,20142.02
> 2048,19627.29,20298.92
> 4096,19668.58,20392.57
> 8192,19844.69,20428.23
> 16384,19786.79,20461.12
> 32768,19735.35,20524.20

The file is not pre-allocated.  Here is the code for the above test.
```

 if (!file.open("TeensyDemo.bin", O_RDWR | O_CREAT)) {
    errorHalt("open failed");
  }
  Serial.println("\nsize,write,read");
  Serial.println("bytes,KB/sec,KB/sec");
  for (size_t nb = 512; nb <= BUF_DIM; nb *= 2) {
    file.truncate(0);
    uint32_t nRdWr = FILE_SIZE/nb;
    Serial.print(nb);
    Serial.print(',');
    uint32_t t = micros();
    for (uint32_t n = 0; n < nRdWr; n++) {
      // Set start and end of buffer.
      buf32[0] = n;
      buf32[nb/4 - 1] = n;
      if (nb != file.write(buf, nb)) {
        errorHalt("write failed");
      }
    }
    t = micros() - t;
    totalMicros += t;
    Serial.print(1000.0*FILE_SIZE/t);
    Serial.print(',');
    file.rewind();
    t = micros();
    
    for (uint32_t n = 0; n < nRdWr; n++) {
      if ((int)nb != file.read(buf, nb)) {
        errorHalt("read failed");
      }
      // crude check of data.     
      if (buf32[0] != n || buf32[nb/4 - 1] != n) {
        errorHalt("data check");
      }
    }
    t = micros() - t;
    totalMicros += t;   
    Serial.println(1000.0*FILE_SIZE/t);    
  }
  file.close();
```

hespanha, 2017-03-29:
Thanks.
Joao

greiman, 2018-12-19:
Closed, this version of SdFat-beta is no longer supported.

closed 2018-12-19 by greiman