greiman/SdFat-beta #16: Card detect in Teensy 3.6

CarlosPosse, 2017-04-19:
When starting using this library with Teensy 3.6 I noted that no Card detect feature was enabled, either through an external pin or through the PRSSTAT register. This is due to the DAT3 line of the SD card (pin PTE4) being pulled high. This is asserted in the enableGPIO function:

`static void enableGPIO(bool enable) {
  const uint32_t PORT_CLK = PORT_PCR_MUX(4) | PORT_PCR_DSE;
  const uint32_t PORT_CMD_DATA = PORT_CLK | PORT_PCR_PS | PORT_PCR_PE;

  PORTE_PCR0 = enable ? PORT_CMD_DATA : 0;  // SDHC_D1
  PORTE_PCR1 = enable ? PORT_CMD_DATA : 0;  // SDHC_D0
  PORTE_PCR2 = enable ? PORT_CLK : 0;       // SDHC_CLK
  PORTE_PCR3 = enable ? PORT_CMD_DATA : 0;  // SDHC_CMD
  PORTE_PCR4 = enable ? PORT_CMD_DATA : 0;  // SDHC_D3
  PORTE_PCR5 = enable ? PORT_CMD_DATA : 0;  // SDHC_D2
}`

Altering the SDHC_D3 line to `PORTE_PCR4 = enable ? (PORT_CLK | PORT_PCR_PE) : 0;` and asserting the D3CD bit of the PROCTL register with `SDHC_PROCTL |= SDHC_PROCTL_D3CD;` enables the card detect feature in the PRSSTAT register, leaving detection to a simple bit check.

Could this bring any unwanted behaviour to the rest of the code ?

greiman, 2017-04-19:
Use of DAT3 is not practical on Teensy 3.6 since there is no external pull-down resistor on DAT3.  The SD internal pull-up is 50K. The Teensy 3.6 internal pull-down is too small to be reliable.

You can try to detect the card before calling sd.begin().   I gave up on using DAT3 on Teensy 3.6 and in general DAT3 is no longer recommended for SD card detect.  Too bad Teensy 3.6 has no card detect switch. 

Also note section 60.1.2 of the K66 data sheet.

> 
> The SD standard requires the SD bus signals (except the SD clock) to be pulled up during
> data transfers. The SDHC also provides a feature of detecting card insertion/removal, by
> detecting voltage level changes on DAT[3] of the SD bus. To support this DAT[3] must
> be pulled down. To avoid a situation where the SDHC detects voltage changes due to
> normal data transfers on the SD bus as card insertion/removal, the interrupt relating to
> this event must be disabled after the card has been inserted and detected. It can be reenabled
> after the card is removed.

greiman, 2018-12-19:
Closed, this version of SdFat-beta is no longer supported.

closed 2018-12-19 by greiman