greiman/SdFat-beta #19: SoftwareSPI Example - Numerous Errors with fastPinConfig

mjs513, 2017-09-28:
Trying to run the SoftwareSPI example sketch throws numerous errors with fastPinConfig.  First, I as for configuration,  I added a new board pin config file for something called flutter wireless it is a SAM3S1A board that is Arduino compatible.  I have used softSPI from your DigitalIO library with not issues - worked fine with no errors with fastPinConfig.   Anyway, listed below are the errors (Pretty much same error repeated for one).  I am missing something?

Thanks
Mike

```
In file included from C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:31:0,

                 from C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SdSpiDriver.h:192,

                 from C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/SdSpiCard.h:30,

                 from C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/BlockDriver.h:27,

                 from C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdFat.h:27,

                 from C:\Users\CYBERP~1\AppData\Local\Temp\arduino_modified_sketch_565098\SoftwareSpi.ino:7:

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:278:25: error: variable or field 'fastPinMode' declared void

 inline void fastPinMode(pin, mode) {

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:278:25: error: 'pin' was not declared in this scope

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:278:30: error: 'mode' was not declared in this scope

 inline void fastPinMode(pin, mode) {

                              ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::config(uint8_t, bool)':

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(PinNumber, mode, level);

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:328:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(PinNumber, mode, level);

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h: In member function 'void DigitalPin<PinNumber>::mode(uint8_t)':

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:352:32: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]

     fastPinMode(PinNumber, mode);

                                ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h: In member function 'void SoftSPI<MisoPin, MosiPin, SckPin, Mode>::begin()':

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:55:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(MisoPin, MISO_MODE, MISO_LEVEL);

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:56:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(MosiPin, MOSI_MODE, !MODE_CPHA(Mode));

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: there are no arguments to 'fastPinMode' that depend on a template parameter, so a declaration of 'fastPinMode' must be available [-fpermissive]

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:57:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(SckPin, SCK_MODE, MODE_CPOL(Mode));

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h: In instantiation of 'void SoftSPI<MisoPin, MosiPin, SckPin, Mode>::begin() [with unsigned char MisoPin = 8u; unsigned char MosiPin = 9u; unsigned char SckPin = 10u; unsigned char Mode = 0u]':

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SdSpiDriver.h:215:5:   required from 'void SdSpiSoftDriver<MisoPin, MosiPin, SckPin>::begin(uint8_t) [with unsigned char MisoPin = 8u; unsigned char MosiPin = 9u; unsigned char SckPin = 10u; uint8_t = unsigned char]'

C:\Users\CYBERP~1\AppData\Local\Temp\arduino_modified_sketch_565098\SoftwareSpi.ino:55:14:   required from here

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: 'fastPinMode' was not declared in this scope

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:55:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(MisoPin, MISO_MODE, MISO_LEVEL);

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: 'fastPinMode' was not declared in this scope

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:56:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(MosiPin, MOSI_MODE, !MODE_CPHA(Mode));

     ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/DigitalPin.h:290:25: error: 'fastPinMode' was not declared in this scope

   {fastPinMode(pin, mode); fastDigitalWrite(pin, level);}

                         ^

C:\Users\CyberPalin\Documents\Arduino\libraries\SdFat\src/SdCard/../SpiDriver/SoftSPI.h:57:5: note: in expansion of macro 'fastPinConfig'

     fastPinConfig(SckPin, SCK_MODE, MODE_CPOL(Mode));
```

greiman, 2017-09-28:
You must add your own code to support SAM3S1A.

mjs513, 2017-09-28:
Actually, what I did was copy the SoftSPI.h and DigitalPins.h files from you DigitalIO library and replaced the existing files in the SDFat library with them.  It compiled and worked fine.  It wrote the file and disc and pasted the text correctly.

You may want to do the same with the beta library.

Have another problem which may be a setting, have to do a little more trouble shooting on that one.

thanks
Mike

closed 2017-09-28 by mjs513

greiman, 2017-09-28:
Sorry I should have marked this library as out of date.  You should be using SdFat, it is most current.

I think SdFat has a newer version of DigitalPins.h and SoftSPI.h.

mjs513, 2017-09-28:
Just replaced the SDfat-beta with SDFAT and got the same error messages as with the SDfat-beta library.  When I replaced the same files as mentioned above it compiled no errors and ran.  Just thought you should know.

Mike

greiman, 2017-09-28:
Looks like this mod didn't get to DigitalPin.h in SdFat.

```
-  inline void fastPinMode(pin, mode) {

+  inline void fastPinMode(uint8_t pin, uint8_t mode) {
```

Should be OK now.

mjs513, 2017-09-29:
Thanks for getting back to me.  I am very happy with your libs they are great.  Thanks for putting these together, saved my life.

Mike