greiman/SdFat-beta #21: New Teensy

mjs513, 2019-06-04:
Hi Bill 
Trying to get new Teensy T$ working on a SPI.  Is there a way to specify what pins to use for SPI1, ie., MISO, MOSI, SCK?  See this thread for the Teensy: https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test.  Works right out of the box for SPI.

greiman, 2019-06-07:
I don't have a Teensy 4.0 so I can't modify code for it.

I didn't ask Paul for a 4.0 beta since I was busy finishing SdFat V 2.0 beta.  Guess I should try to get a Teensy 4.0.

Here is the section of SdFatConfig.h where SPI pins are defined for Teensy 3.5/3.6.
```
/** Enable SDIO driver if available. */
#if defined(__MK64FX512__) || defined(__MK66FX1M0__)
// Pseudo pin select for SDIO.
#ifndef BUILTIN_SDCARD
#define BUILTIN_SDCARD 254
#endif  // BUILTIN_SDCARD
// SPI for built-in card.
#ifndef SDCARD_SPI
#define SDCARD_SPI      SPI1
#define SDCARD_MISO_PIN 59
#define SDCARD_MOSI_PIN 61
#define SDCARD_SCK_PIN  60
#define SDCARD_SS_PIN   62
#endif  // SDCARD_SPI
#define HAS_SDIO_CLASS 1
#endif  // defined(__MK64FX512__) || defined(__MK66FX1M0__)
```

Here is the section of SdSpiTeensy3.cpp where the definitions are used.
```
void SdAltSpiDriver::begin(SdSpiConfig spiConfig) {
  m_csPin = spiConfig.csPin;
  m_spiSettings = SPI_LOW_SPEED;
  if (spiConfig.spiPort) {
    m_spi = spiConfig.spiPort;
#if defined(SDCARD_SPI) && defined(SDCARD_SS_PIN)
  } else if (m_csPin == SDCARD_SS_PIN) {
    m_spi = &SDCARD_SPI;
    m_spi->setMISO(SDCARD_MISO_PIN);
    m_spi->setMOSI(SDCARD_MOSI_PIN);
    m_spi->setSCK(SDCARD_SCK_PIN);
#endif  // defined(SDCARD_SPI) && defined(SDCARD_SS_PIN)
  } else {
    m_spi = &SPI;
  }
  pinMode(m_csPin, OUTPUT);
  digitalWrite(m_csPin, HIGH);
  m_spi->begin();
}
```

mjs513, 2019-06-07:
Thanks BIll… I looked at that but couldn't figure out how it knew to use SdSpiTeensy3.cpp.

Anyway did a hack to SDFatConfig.h:
```
//------------------------------------------------------------------------------
/** Enable SDIO driver if available. */
#if defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__IMXRT1062__)
// Pseudo pin select for SDIO.
#ifndef BUILTIN_SDCARD
#define BUILTIN_SDCARD 254
#endif  // BUILTIN_SDCARD
// SPI for built-in card.
#if defined(__IMXRT1062__)
	#ifndef SDCARD_SPI
	#define SDCARD_SPI      SPI1
	#define SDCARD_MISO_PIN 34
	#define SDCARD_MOSI_PIN 35
	#define SDCARD_SCK_PIN  37
	#define SDCARD_SS_PIN   36
	#endif  // SDCARD_SPI
	#define HAS_SDIO_CLASS 0
#else
	#ifndef SDCARD_SPI
	#define SDCARD_SPI      SPI1
	#define SDCARD_MISO_PIN 59
	#define SDCARD_MOSI_PIN 61
	#define SDCARD_SCK_PIN  60
	#define SDCARD_SS_PIN   62
	#endif  // SDCARD_SPI
	#define HAS_SDIO_CLASS 1
#endif
#endif  // defined(__MK64FX512__) || defined(__MK66FX1M0__)
```
But still didn't work:
```
SdFat version: 2.0.0

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 36
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start

SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is there a wiring/soldering problem?
Is SD_CS_PIN set to the correct value?
Does another SPI device need to be disabled?
SD errorCode: SD_CARD_ERROR_CMD0 = 0x1
SD errorData = 0x0

type any character to start

```  So something else may be going on so I will leave it up to you..... :)  Thanks anyway, I thought there might be a simple workaround but I guess not.

closed 2021-10-01 by mjs513