greiman/SdFat-beta #22: Beta2 - Excellent !!

gwevans22, 2019-06-07:
Bill:
Personal thanks for all your work in developing SdFat - without which many uMakers worldwide would not be using sd cards!  Needed your Beta 2 to finally get a my logger working on Teensy 3.6 with 2 I2C buses and RTC.  Look forward to updated SdInfo that works with Sdio and ExFat.
Thanks again.
Gary Evans - gary.evans@uhn.ca

greiman, 2019-06-07:
Did you try this version of [SdInfo](https://github.com/greiman/SdFat-beta/tree/master/examples/SdInfo)?

I added the following two line to do an ls and ran it on a Teensy 3.6 with an exFAT sd that had been used on an Android phone.

```
  dmpVol();
  cout << F("\nls:") <<endl;  // added
  sd.ls();  // added
```
Here is the result:
```
SdFat version: 2.0.0
Assuming an SDIO interface.

type any character to start
init time: 8 ms

Card type: SDXC

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X2C06AFA0
Manufacturing date: 9/2015

cardSize: 64021.86 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,125009920
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is exFAT
sectorsPerCluster: 256
clusterCount:      488192
freeClusterCount:  488032
fatStartSector:    49152
dataStartSector:   65536

ls:
bench.dat
LOST.DIR/
.android_secure/
Android/
TeensyDemo.bin
RtcTest.txt
Dir1/
```

gwevans22, 2019-06-10:
Bill:  Your link works fine!  Don't know how, but the version *supposedly*
loaded into my IDE, (taken from the Beta 2 zip file downloaded from
GitHub), is not the same as your link, (still has #include <SPI.h>).
Problem obviously here.  I'll check some other example files to make sure
they're actually from your Beta 2.  Thank you again for your prodigious
works!
gary