greiman/SdFat-beta #23: Sd card not found on Mega2560

drmpf, 2019-06-17:
I am trying your great AVR logger example.
Runs on Uno but when I move to Mega (using pins 50 to 53 for SPI)
#define PIN_SPI_SS    (53)
#define PIN_SPI_MOSI  (51)
#define PIN_SPI_MISO  (50)
#define PIN_SPI_SCK   (52)
 it fails to find the sd card
Any suggestions for debugging further?

greiman, 2019-06-17:
I connected an [Adafruit MicroSD breakout](https://www.adafruit.com/product/254) to these pins and ran the AvrAdcLogger example without modification.

Here is the output:
```

Type any character to begin.

Unused stack: 6923

type:
b - open existing bin file
c - convert file to csv
l - list files
p - print data to Serial
r - record ADC data

Opening: AvrAdc00.bin
Allocating: 100 MiB
Sample pins: 0 1 2 3 4
ADC bits: 10
ADC clock kHz: 500
Sample Rate: 5000.00
Sample interval usec: 200.00
Logging - type any character to stop

Truncating file
Max write latency usec: 1068
Record time sec: 12.306
Sample count: 61530
Overruns: 0
FIFO_DIM: 96
maxFifoUse: 2
Done
```

drmpf, 2019-06-17:
Odd, I am using the same Adafruit breakout board.  
I will try again.

drmpf, 2019-06-20:
OK found the problem.
You need to connect MOSI -> DI and MISO -> DO on adafruit board.

closed 2019-06-20 by drmpf