greiman/SdFat-beta #28: createContiguous not available in ExFile and FsFile

ccdzapper, 2019-11-18:
Hi Bill,
Is there a way to create a contiguous file if I declare a file of type ExFile or FsFile ? 
I am trying to use this in a low-latency application modeled on the original LowLatencyLogger (which, BTW I do not see in the SD Beta Examples folder).
I am getting the following error:
**SDDirectTest.ino: 126:15: error: 'class FsFile' has no member named 'createContiguous
   if (!binFile.createContiguous(TMP_FILE_NAME, 512 * FILE_BLOCK_COUNT))**

Thank You,
Ram

greiman, 2019-11-19:
The following works for all file types.

Create an empty file with some form of open or truncate an existing file.

Then call preAllocate(size).  Here is the call in the bench example at about line 190.

```
    file.truncate(0);
    if (PRE_ALLOCATE) {
      if (!file.preAllocate(FILE_SIZE)) {
        error("preAllocate failed");
      }
    }
```

ccdzapper, 2019-11-29:
Bill,

Thank you. I did get a low latency application to work really well with dedicated SPI. When I tried to adapt it to SDIO using either FIFO_MODE or DMA_MODE, I found that the file writes appear to be blocking calls - as a result during the occasional long SD writes of 50 mSecs when the card is off doing it's own thing isBusy() always returns false after the write is complete. ( With dedicated SPI, this is not the case).

After reading a lot of forum discussions on Teensy 3.6 and 4.0 (mainly between you and defragster) I thought maybe I could use yield() to stack up sensor data during the busy times, but again, I registered 0 yield calls during writes.

Just to be sure, I modified your TeensySDIODemo program to accumulate the yield stats separately for reads and writes and here's the result I got: (Teensy 3.6)

FIFO SDIO mode.

size,write,read
bytes,KB/sec,KB/sec
512,19249.07,18945.62
1024,18976.90,19104.53
2048,19614.81,19293.52
4096,19712.48,19388.95
8192,19692.63,19437.69
16384,19709.80,19463.17
32768,19834.79,19473.34

totalMicros  6048196
write yieldMicros  0
write yieldCalls   0
write yieldMaxUsec 0
--------------
read yieldMicros  37281
read yieldCalls   31
read yieldMaxUsec 4381
Done

Type '1' for FIFO SDIO
     '2' for DMA SDIO
     '3' for Dedicated SPI
     '4' for Shared SPI

Does this mean that only dedicated SPI can be used for low latency logging ?

Thank you, and BTW, thanks for your generous contributions to the embedded systems community.

greiman, 2019-11-29:
Busy for SDIO is not reliable on Teensy.  I have tried to find the problem without success.

ccdzapper, 2019-12-24:
Is there any other SDIO implemented Arduino platform that you have tested that does support Busy ?
I guess I am greedy to use the both the high data rates and the low latency (non-blocking sector writes).

greiman, 2019-12-24:
There is no other Platform I know of.  Part of the problem is that a busy test doesn't help on most platforms since you need the ability to do infinite transfers with SDIO.   When you use DMA on most platforms, you can't control wtire latency.

Teensy 3.0 allows very large transfers in FIFO mode but has a bug that will cause busy to fail.

I plan to look at Teensy 4.0 again.

iharshadec, 2020-03-25:
> There is no other Platform I know of. Part of the problem is that a busy test doesn't help on most platforms since you need the ability to do infinite transfers with SDIO. When you use DMA on most [...]

Not sure if the same issue but yes with Teensy 4.0 I see sometimes SD write is getting failed. I see there is latest commit in Feb 2020. Any of this Teensy 4.0 issue get fixed?

greiman, 2020-03-25:
I don't have an example of write failures on Teensy 4.0.  I have two test systems, one Paul sent me with a large breakout board and one that I added a ribbon connector and SD socket.

The write busy sense is not use in a normal write.  I is intended to avoid writes when the SD is busy programming flash.