greiman/SdFat-beta #31: Reporting erased state of card

gbhug5a, 2020-01-25:
Would you consider reporting the value of the DATA_STAT_AFTER_ERASE bit (bit 55) of the SCR register in SDInfo.ino?  This is the erased state of the flash memory, which can differ among manufacturers.  This would require adding support for ACMD51 to the library source, or possibly by erasing a block and reading back the first byte.

greiman, 2020-01-25:
I put this on my to do list.   I need to finish a set of features that allow USB drives and more flexible software SPI. 

You can determine the DATA_STAT_AFTER_ERASE bit by running the SdFormatter example.  Select the E or F option and the erase value will be printed.

```
Enter option: F

Erasing
................................................................
................................................................
................................................................
................................................................
................................................................
................................................................
................................................................
................................................
All data set to 0x00
```

gbhug5a, 2020-01-25:
Very good.  Thanks very much.

closed 2020-01-25 by gbhug5a