greiman/SdFat-beta #32: experience with SdFat-beta in Sloeber - ArduinoIDE Plug-in

Sebastiangka, 2020-01-28:
Hello Sir,
I'm having problems trying to run the examples programs from this library in Sloeber. I don´t know if someone else had this already. 

I also noticed that for FreeStack is for Sloeber not defined. Which are the consequences if it's not defined? I´m wondering if that's the reason why it doesn't work...

greiman, 2020-01-28:
It mean none of the required defines are set for your board in Soleber so you will get zero if you call FreeStack.  FreeStack is not used in the SdFat code.

I only test with the official Arduino IDE.  I don't have time to test with every IDE and Arduino like board.

Sebastiangka, 2020-01-29:
In case someone want to know about this, it´s compatible/runable to use this beta version on Sloeber.
For more Infos check this [https://github.com/Sloeber/arduino-eclipse-plugin/issues/1137](url)

closed 2020-01-29 by Sebastiangka