greiman/SdFat-beta #33: writing block-wise in sd from an array type struct

Sebastiangka, 2020-02-10:
### Hardware:
Board:							NodeMCU-32S
IDE name:						Sloeber
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                          Windows 10

### Description:
Hi Mr. Greiman,
I'm using the Beta Version to save data from a sensor into an sd card. I defined an array type "sample", which is a structure with 32 Byte size. Since my µC only has 600µs to write in between I decided to write in 1024 Byte-Blocks, perhaps my array´s size 4096 Bytes is. After running the program and checking the sd, I saw completely different values than I expected so I'm doing something wrong and I suppose it´s my approach with the buffer pointer.
I thought about casting my values as a string but this takes to long.
Which is the popper way to approach in my case?
Thank you very much.
Sebastian

```cpp
#include <Arduino.h>
#include "SdFat.h"
#include "sdios.h"
#define SD_FAT_TYPE 0
#define SPI_CLOCK SD_SCK_MHZ(20)
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = SS;
#endif  // SDCARD_SS_PIN
#if ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#endif  // HAS_SDIO_CLASS
// Set PRE_ALLOCATE true to pre-allocate file clusters.
const bool PRE_ALLOCATE = true;
bool skipLatency = false;
#if SD_FAT_TYPE == 0
SdFat sd_race;
File file_race; // @suppress("Abstract class cannot be instantiated")
#endif
const size_t BUF_SIZE = 512;
const uint32_t FILE_SIZE_MB = 1;
const uint32_t FILE_SIZE = 1000000UL*FILE_SIZE_MB;
//---------------------------------
struct sampleacc{				//32Bytes
	float x;	char tab ='\t';
	float y;	char tab2='\t';
	unsigned long stamp;
	bool stateIE;
};
sampleacc dataToSd[BUF_SIZE]; //4096Bytes	
uint8_t* buffer = (uint8_t*)dataToSd;
const size_t WBUF_SIZE = 1024;
const size_t WRITE_GAP = 32;
int16_t idxRData = 0;
int16_t idxWData = 0;
unsigned long tick;
//-----------------------------------------------------
void setup() {
Serial.begin(BaudRate);
	  while (!Serial) {
	    SysCall::yield();
	  }
	do{
	    delay(10);
	} while (Serial.available() && Serial.read() >= 0);
	if (!sd_race.begin(SD_CONFIG)) {
	    sd_race.initErrorHalt(&Serial);
	 }
	if (!file_race.open("race1.dat", O_RDWR | O_CREAT | O_TRUNC)) {
			error("open failed");}
}

void loop() {
dataToSd[idxRData].stamp = tick;     
dataToSd[idxRData].x = (float)77;    //just to try    
dataToSd[idxRData].y = (float)66;
idxRData++;
if(abs(idxRData-idxWData) >= WRITE_GAP){
	if(idxRData-idxWData <= 0){
		Serial.println("idxW write till end of array");
		buffer =(uint8_t*) &dataToSd[idxWData];
      	if (file_race.write(buffer, (WBUF_SIZE-idxWData -1)) != (WBUF_SIZE-idxWData-1)) {
				  error("write failed");
		}
		idxWData = 0;
	}
	else{
		Serial.println("Values will be writen it in Sd ");
		buffer =(uint8_t*) &dataToSd[idxRData];
		if (file_race.write(buffer, WBUF_SIZE) != WBUF_SIZE) {
			  error("write failed");
                }
	        if (skipLatency) {  // Wait until first write to SD, not just a copy to the cache.
				  skipLatency = file_race.curPosition() < WBUF_SIZE;
		  } else {file_race.sync();}
		idxWData=idxRData;
		}
	}
if(idxRData == (BUF_SIZE - 1)){
	Serial.println("Array overflow");
	idxRData = 0;
}
Serial.print(idxRData); Serial.print("\t");
Serial.println(idxWData);
if(file_race.fileSize()>=FILE_SIZE){file_race.close();Serial.println("File is full");while(1){;}}
}
```

greiman, 2020-02-10:
I can't help you write apps, too many people ask.

If you find a bug, isolate the bug in your app and post a simple example that demonstrates the bug.

Sebastiangka, 2020-02-10:
In the picture bellow you can see what it comes out after running the example program bench with some modifications.
[image]
here is the code i added to represent in a simple way my case.
```c
struct sample{
  float x; float y;
  };
sample lino[BUF_SIZE];
sample* pl = lino;
.
.
.
  // fill buf with known data
  if (BUF_SIZE > 1) {
    for (size_t i = 0; i < (BUF_SIZE - 2); i++) {
      buf[i] = 'A' + (i % 26);
      pl[i].x = esp_random();
      Serial.print(pl[i].x);
      Serial.print('\t');
      pl[i].y = esp_random();
      Serial.println(pl[i].y);
    }
    buf[BUF_SIZE-2] = '\r';
    pl[BUF_SIZE-2].x = 0.0;
    pl[BUF_SIZE-2].y = 0.0;
  }
  buf[BUF_SIZE-1] = '\n';
  pl[BUF_SIZE-1].x = 0.0;
  pl[BUF_SIZE-1].y = 0.0;

```
The size from the bench file is correct:
[image]

I would like to know if there is a special reason why you define buf as a uint_8 pointer and cast buffer32 into an int8_t.

Sebastiangka, 2020-02-11:
I toked a look at the FatFile write function, then I had to inform my self about "reinterpret_cast conversion" and so far as I understand it works only for integers and pointers. As you defined in write-function, buf is void* pointer type and it will be casted direct at the beginning.
```c
const uint8_t* src = reinterpret_cast<const uint8_t*>(buf);
```
I would like to know how to work with different variables types like float and struct, in order to write them in the Sd-Card. Or if its actually possible to work with them.
Thanks for your time

greiman, 2020-02-11:
```
if (file.write(pl, BUF_SIZE) != BUF_SIZE) {
        error("write failed");
      }
```
The size of pl is not BUF_SIZE bytes.  It is sizeof(pl) so you are only writing a portion of the array.

Casting a variable will not result in a text file. 

I cast a uint32_t to uint8_t in the bench example since some processors require DMA writes to be aligned on 4-byte boundaries for maximum speed.  The Teensy 4 board can write at 22 MB/sec with it's fast 4-bit wide SDHC  controller.

> Since my µC only has 600µs to write

This rate can't be achieved with a simple program.  SD cards can have write latency times of many milliseconds so you must acquire and buffer data to write at this speed.  This may require data acquisition in an interrupt routine or a RTOS.

It may not be possible with your board since it is slow for SPI writes.

Sebastiangka, 2020-02-11:
> This may require data acquisition in an interrupt routine or a RTOS.
That´s my plan. I use a time interrupt to check the values from my sensor, this happen every 1ms and it´s needs ~350µs to read the values. So every 1ms it will be a new sample to my array, is the array full then should be copied in the Sd.

I would be satisfied if I get at least the right values in the Sd, then i would work on the timing problem.

greiman, 2020-02-11:
You must fix you bugs.  You are not writing the correct amount of data to the SD.  The data is written as binary, not text.

I am closing this since I don't have time to tutor C++ programming.

closed 2020-02-11 by greiman