greiman/SdFat-beta #35: Error compiling for board Arduino Nano 33 ble

sachamarbacher, 2020-03-22:
i tried to use the library with the folowing hardware:
- arduino Nano 33 BLE
- adafruit addaloger feather
if i use the standard SD library for to write or read to the sd card everything is working fine.
now i nee longer file name then 8.3.
but everytime i try to compile every sketch example out of the SdFat-beta i get the same error:

Please help me...

in these exemple i used the sketch "Sdino"

Error Message:

Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Nano 33 BLE"

In file included from d:\documents\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\_default_fcntl.h:200:0,

                 from d:\documents\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\fcntl.h:4,

                 from d:\documents\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\fcntl.h:1,

                 from d:\documents\arduino\libraries\sdfat\src\common\fsapiconstants.h:30,

                 from D:\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatFile.h:36,

                 from D:\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatVolume.h:28,

                 from D:\Documents\Arduino\libraries\SdFat\src/ExFatLib/ExFatLib.h:27,

                 from D:\Documents\Arduino\libraries\SdFat\src/SdFat.h:33,

                 from C:\Users\Sacha\AppData\Local\Temp\arduino_modified_sketch_98204\SdInfo.ino:4:

d:\documents\arduinodata\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\stat.h:27:8: error: redefinition of 'struct stat'

 struct stat

        ^~~~

In file included from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_error.h:21:0,

                 from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Queue.h:28,

                 from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/Mail.h:28,

                 from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/rtos/rtos.h:33,

                 from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/mbed.h:23,

                 from D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/Arduino.h:32,

                 from sketch\SdInfo.ino.cpp:1:

D:\Documents\ArduinoData\packages\arduino\hardware\mbed\1.1.4\cores\arduino/mbed/platform/mbed_retarget.h:500:8: note: previous definition of 'struct stat'

 struct stat {

        ^~~~

exit status 1
Error compiling for board Arduino Nano 33 BLE.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

greiman, 2020-03-22:
Looks like the Nano 33 BLE board package has a bug for use of fcntl.h

This sketch gives the same error.
```
#include <fcntl.h>
void setup() {
}
void loop() {
}
```
SdFat-beta must use this file to avoid problems with other boards.  You can try editing SdFat/src/SdFatConfig.h to not use fcntl.h.

At about line 195 set USE_FCNTL_H to zero like this.

```
#elif defined(__arm__)
// ARM gcc defines open flags.
#define USE_FCNTL_H 0  // <<-- was 1 
#elif defined(ESP32)
```
This seems to eliminate the compile error but may cause other problems.  I don't have a Nano 33 BLE to test.

sachamarbacher, 2020-03-22:
Hi Greiman

Thank you very much!
I just Updated the SdFatConfig.h file like you told and now the example works perfect.

Greetings
Sacha

closed 2020-03-22 by sachamarbacher

arduinomnomnom, 2020-03-27:
I encountered the same error on a Nano 33 BLE Sense. Found this, which solved the compile problem for me:

https://github.com/arduino/ArduinoCore-nRF528x-mbedos/issues/49

https://github.com/ARMmbed/mbed-os/pull/12607/files