greiman/SdFat-beta #36: benchmarks for Nano 33 BLE Sense

arduinomnomnom, 2020-03-27:
Running "bench" on a Nano 33 BLE Sense.
("Struct Stat" compile error error fixed using [https://github.com/ARMmbed/mbed-os/pull/12607/files](url))

I am wondering if 46.84 KB/Sec write and 47.28 KB/Sec read is the best this board can do, due to the implementation on top of MBED. (See [https://forum.arduino.cc/index.php?topic=642496.0](url))

 I tried with Toshiba and a SandDisk 16GB SD Cards.

Benchmark output:

> FreeStack: -5092
> Type is FAT32
> Card size: 15.55 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X2
> OEM ID: TM
> Product: SA16G
> Version: 3.1
> Serial number: 0X30279627
> Manufacturing date: 10/2011
> 
> FILE_SIZE_MB = 1
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 46.84,11017,10864,10921
> 46.85,11016,10864,10920
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 47.28,10865,10742,10819
> 47.25,10865,10742,10828

greiman, 2020-03-27:
> I am wondering if 46.84 KB/Sec write and 47.28 KB/Sec read is the best this board can do, due to the implementation on top of MBED.

I looked at the NANO 33 BLE SPI library. It will be very slow as implemented.

It might be possible to increase the speed but will be tricky and might not work since it would require direct access to the MBED SPI driver, not the Arduino wrapper.

I have ordered a NANO 33 BLE and will see what is possible.

arduinomnomnom, 2020-03-29:
Thanks for looking into this. 
And thank you for a great library; I've had the pleasure of using it successfully on a variety of AVR boards.

gnodipac886, 2021-07-30:
Hi @greiman 

I was wondering if there were any discoveries that can improve the sd card performance for this board?

Thanks in advance!

greiman, 2021-07-30:
You could try editing SdFatConfig.h and set USE_SPI_ARRAY_TRANSFER  to one in this section.
```
/**
 * If USE_SPI_ARRAY_TRANSFER is non-zero and the standard SPI library is
 * use, the array transfer function, transfer(buf, size), will be used.
 * This option will allocate up to a 512 byte temporary buffer for send.
 * This may be faster for some boards.  Do not use this with AVR boards.
 */
#ifndef USE_SPI_ARRAY_TRANSFER
#define USE_SPI_ARRAY_TRANSFER 0
#endif  // USE_SPI_ARRAY_TRANSFER
```

arduinomnomnom, 2021-07-31:
By coincidence, I was recently reviewing my project from last year. Using fresh copies of most of the libraries, I was wondering why a small SD card write was benching at about 130ms. Changing USE_SPI_ARRAY_TRANSFER to 1 reduced the time to 25~30ms. So, thanks for the confirmation.
I'm not aware of any improvements in the Arduino SPI implementation on top of MBED; I get the impression online that there isn't must love for the Nano 33 BLE; the RP2040 seems to be where the interest is.

greiman, 2021-07-31:
The RP2040 is great for a $4 board.  I get about 2 MB/sec with SdFat using array transfer.  Too bad its ADC has a major fault.  Only about 8.5 bits accuracy.  Hope the ADC is corrected with new masks to remove the errata.  I used the second core and it is nice to have two threads.

I have the Arduino Nano version but have not had time to play with it.

arduinomnomnom, 2021-08-01:
There seems to be official acknowledgement of the RP2040 issue, so it looks likely that there will be a fix in future versions. I'll definitely explore them then. I've looked at the ESP32, but that apparently has a noisy ADC. The search for perfection continues...

arduinomnomnom, 2021-08-02:
Here are the results running bench using SDFAT 2.0.7 on the Arduino Nano 33 BLE Sense with USE_SPI_ARRAY_TRANSFER 1 and FILE_SIZE_MB = 1. Looking a lot better than my first post above.

```
FreeStack: -6064
Type is FAT32
Card size: 15.55 GB (GB = 1E9 bytes)

Manufacturer ID: 0X2
OEM ID: TM
Product: SA16G
Version: 3.1
Serial number: 0X30279627
Manufacturing date: 10/2011

FILE_SIZE_MB = 1
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
408.50,127134,1165,1247
430.48,1306,1165,1182

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
459.56,1139,1093,1107
459.56,1139,1093,1107

```

greiman, 2021-08-02:
Still less than an Uno but acceptable for many applications.

Uno:
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 644.66,107836,728,788
> 678.79,2584,728,748
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 671.14,768,752,756
> 671.14,768,752,756