greiman/SdFat-beta #39: read/write failures on new T4.1 builtin SD card

manitou48, 2020-04-23:
On T4.1 builtin  SD card, reading a file would  terminate with an error (BDE BCE bits set).  Sketch with same uSD card would work on T4.1 with teensy SD lib, and it would work on T4.0 with SdFat-beta.
Fix was to change  DSE(4) back to DSE(1) SdioTeensy.cpp.   Not sure if that breaks something else, or why you changed it to DSE(4).  additional testing ongoing ...

greiman, 2020-04-23:
I don't recall why the mod was made and don't have time to pursue this.

manitou48, 2020-04-24:
OK, defragster pointed out that DSE(4) is still required for the Teensy 4.0 breakout boards.  So a conditional fix to SdioTeensy.cpp is needed, 
```
static void enableGPIO(bool enable) {
  const uint32_t CLOCK_MASK = IOMUXC_SW_PAD_CTL_PAD_PKE |
#if defined(ARDUINO_TEENSY41)
                              IOMUXC_SW_PAD_CTL_PAD_DSE(1) |
#else
                              IOMUXC_SW_PAD_CTL_PAD_DSE(4) |  ///// WHG
#endif
                              IOMUXC_SW_PAD_CTL_PAD_SPEED(2);

  const uint32_t DATA_MASK = CLOCK_MASK | IOMUXC_SW_PAD_CTL_PAD_PUE |
                             IOMUXC_SW_PAD_CTL_PAD_PUS(1);

```

greiman, 2020-04-24:
I made a new release with your fix.  Please test it, I don't have a T4.1.

manitou48, 2020-04-24:
Looks good. I tested on T4.1 and T4.0. thanks
you are on the list for a free T4.1, see
https://forum.pjrc.com/threads/60532-Teensy-4-1-Beta-Test

closed 2020-04-24 by manitou48