greiman/SdFat-beta #40: SD_CARD_ERROR_CMD0 - Card reset failed

williamesp2015, 2020-04-23:
I am using ESP32 and I modified SPI clock to 10 Mhz but i have problem with sd.begin(SD_CONFIG)

williamesp2015, 2020-04-24:
Finally I solved the problem with ugly way.
first change
#define SPI_CLOCK SD_SCK_MHZ(25)
second change
add this configuration to sdfatconfig.h after ESP32
#define SDCARD_MISO_PIN 5//19
#define SDCARD_MOSI_PIN 19//23
#define SDCARD_SCK_PIN 18// 18
#define SDCARD_SS_PIN 23//5
finally in SdSpiESP.cpp
void SdSpiArduinoDriver::begin(SdSpiConfig spiConfig) {
  (void)spiConfig;
  //SPI.begin();
SPI.begin(SDCARD_SCK_PIN,SDCARD_MISO_PIN,SDCARD_MOSI_PIN,SDCARD_SS_PIN);//CLK_PIN,SO_PIN,SI_PIN,CS_PIN
}
please force SdFat Beta to use ESP32 really custom SPI

greiman, 2020-04-24:
The ESP32 has a bug that requires explicit setting of 25 MHz. The spec for SPI::beginTransaction() is to use the highest supported SPI clock less than or equal to the requested value.

I can't change the default pins for existing code.  The defaults in the SPI driver are:

```
void setup() {
  Serial.begin(9600);
  // Wait for USB Serial
  while (!Serial) {
    yield();
  }  
  Serial.println();
  Serial.print("SS ");Serial.println(SS);
  Serial.print("SCK ");Serial.println(SCK);
  Serial.print("MISO ");Serial.println(MISO);
  Serial.print("MOSI ");Serial.println(MOSI);
}
void loop() {
}
```
```
output:
SS 5
SCK 18
MISO 19
MOSI 23
```

williamesp2015, 2020-04-24:
@grieman, you defined custom SPI but did not use it for ESP32&ESP8266. Simply can do as I did.

greiman, 2020-04-24:
Custom SPI is define at about line 314 of SdFatConfig.h
```
#if defined(ARDUINO_ARCH_APOLLO3)\
  || defined(__AVR__)\
  || defined(ESP8266) || defined(ESP32)\
  || defined(PLATFORM_ID)\
  || defined(ARDUINO_SAM_DUE)\
  || defined(__STM32F1__) || defined(__STM32F4__)\
  || (defined(CORE_TEENSY) && defined(__arm__))
#define SD_HAS_CUSTOM_SPI 1  <<-- HERE
```
It uses SdSpiESP.cpp which has been optimized for ESP8266 and ESP32.

greiman, 2020-04-24:
I will not modify SdSpiESP.cpp.

closed 2020-04-24 by greiman