greiman/SdFat-beta #42: Error when compiling on Arduino Nano Every

Tofer89, 2020-04-29:
**Sketch:**

#include <SdFat.h>

**Target:**

Arduino Nano Every

**Dump:**
Arduino:` 1.8.12 (Windows 10), Board: "Arduino Nano Every, None (ATMEGA4809)"

In file included from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiArduinoDriver.h:94:0,

                 from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\sdspidriver.h:134,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdSpiCard.h:35,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdCard.h:28,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdFat.h:32,

                 from C:\Users\Christofer\Desktop\Arduino\Einzelbeispiele\Test\Test.ino:2:

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h: In member function 'uint8_t SdSpiArduinoDriver::receive(uint8_t*, size_t)':

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:52:3: error: 'SPDR' was not declared in this scope

   SPDR = 0XFF;

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:52:3: note: suggested alternative: 'SPI'

   SPDR = 0XFF;

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:54:14: error: 'SPSR' was not declared in this scope

     while (!(SPSR & _BV(SPIF))) {}

              ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:54:14: note: suggested alternative: 'SPI'

     while (!(SPSR & _BV(SPIF))) {}

              ^~~~

              SPI

In file included from c:\users\christofer\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0,

                 from c:\users\christofer\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\pgmspace.h:90,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/String.h:31,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Print.h:24,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Stream.h:25,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Client.h:22,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/ArduinoAPI.h:29,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/Arduino.h:23,

                 from sketch\Test.ino.cpp:1:

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:54:25: error: 'SPIF' was not declared in this scope

     while (!(SPSR & _BV(SPIF))) {}

                         ^

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:54:25: note: suggested alternative: 'SPI'

In file included from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiArduinoDriver.h:94:0,

                 from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\sdspidriver.h:134,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdSpiCard.h:35,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdCard.h:28,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdFat.h:32,

                 from C:\Users\Christofer\Desktop\Arduino\Einzelbeispiele\Test\Test.ino:2:

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:62:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & _BV(SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:62:12: note: suggested alternative: 'SPI'

   while (!(SPSR & _BV(SPIF))) {}

            ^~~~

            SPI

In file included from c:\users\christofer\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0,

                 from c:\users\christofer\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\pgmspace.h:90,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/String.h:31,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Print.h:24,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Stream.h:25,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/Client.h:22,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/api/ArduinoAPI.h:29,

                 from C:\Users\Christofer\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6\cores\arduino/Arduino.h:23,

                 from sketch\Test.ino.cpp:1:

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:62:23: error: 'SPIF' was not declared in this scope

   while (!(SPSR & _BV(SPIF))) {}

                       ^

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:62:23: note: suggested alternative: 'SPI'

In file included from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiArduinoDriver.h:94:0,

                 from c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\sdspidriver.h:134,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdSpiCard.h:35,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdCard/SdCard.h:28,

                 from C:\Users\Christofer\Documents\Arduino\libraries\SdFat-beta-master\src/SdFat.h:32,

                 from C:\Users\Christofer\Desktop\Arduino\Einzelbeispiele\Test\Test.ino:2:

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h: In member function 'void SdSpiArduinoDriver::send(const uint8_t*, size_t)':

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:75:3: error: 'SPDR' was not declared in this scope

   SPDR = *buf++;

   ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:75:3: note: suggested alternative: 'SPI'

   SPDR = *buf++;

   ^~~~

   SPI

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:78:14: error: 'SPSR' was not declared in this scope

     while (!(SPSR & (1 << SPIF))) {}

              ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:78:14: note: suggested alternative: 'SPI'

     while (!(SPSR & (1 << SPIF))) {}

              ^~~~

              SPI

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:78:27: error: 'SPIF' was not declared in this scope

     while (!(SPSR & (1 << SPIF))) {}

                           ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:78:27: note: suggested alternative: 'SPI'

     while (!(SPSR & (1 << SPIF))) {}

                           ^~~~

                           SPI

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:84:12: error: 'SPSR' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:84:12: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

            ^~~~

            SPI

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:84:25: error: 'SPIF' was not declared in this scope

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

c:\users\christofer\documents\arduino\libraries\sdfat-beta-master\src\spidriver\SdSpiAvr.h:84:25: note: suggested alternative: 'SPI'

   while (!(SPSR & (1 << SPIF))) {}

                         ^~~~

                         SPI

exit status 1
Error compiling for board Arduino Nano Every.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

greiman, 2020-04-29:
Try editing SdFatConfig.h at about line 109 and set SPI_DRIVER_SELECT to 1 like this.

`#define SPI_DRIVER_SELECT 1 //  0`

This should at least allow the sketch to compile.

I don't have a Nano Every and can't support all of the many Arduino compatible boards. 

I no longer have time to support SdFat.

Tofer89, 2020-04-30:
Thank you very much. This worked to fix the compiling issue for Sdfat only..

Unfortunately I get more problem when compiling the source code:
`#include` `<Adafruit_ImageReader.h>`

I think that beta version of SDfat is not designed for usage in Adafruit_ImageReader.h.
Is there a way to do the change you suggested also in SdFat Adafruit-Fork 1.2.3 libary?
There is no SPI_DRIVER_SELECT in the SdFatConfig.h file.

greiman, 2020-04-30:
Can't help with Adafruit version.

closed 2020-04-30 by greiman