greiman/SdFat-beta #43: SDFat doesn't work with the audio library

jonbro, 2020-04-30:
On
Arduino 1.8.12
Teensyduino 1.51

putting the following lines of code in:
```
#include <Audio.h>
#include "SdFat.h"
```

Causes a pretty big wall of errors to get dumped out, here are the first few:
```
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\utility\SdFile.cpp:25:57: error: 'void (* SdFile::dateTime_)(uint16_t*, uint16_t*)' is not a static data member of 'class SdFile'

 void (*SdFile::dateTime_)(uint16_t* date, uint16_t* time) = NULL;

                                                         ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\utility\SdFile.cpp:33:28: error: no 'uint8_t SdFile::addCluster()' member function declared in class 'SdFile'

 uint8_t SdFile::addCluster() {

                            ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\utility\SdFile.cpp:46:35: error: no 'uint8_t SdFile::addDirCluster()' member function declared in class 'SdFile'

 uint8_t SdFile::addDirCluster(void) {
```

It seems like there is some kind of conflict between the built in sd lib, and this one.
I confirmed that the library was working without the audio.h include, so I'm pretty sure thats where the conflict is coming from.

FrankBoesing, 2020-04-30:
Yes that's a known issue of the Audio Library - not a problem of SDFat.
Could have been found searching the PJRC Forum ;)

jonbro, 2020-04-30:
ok, my bad, found it:
https://forum.pjrc.com/threads/48410-Including-any-file-in-Audio-library-causes-entire-Audio-library-to-be-included

It still strikes me that it would be possible to resolve this name collision on the side of this library, rather than having ifdefs built into the audio library that detect if this library is being included.

Since we are in c++, would it make sense to namespace this library? Alternately, if you want to keep things clean from the user perspective, you could have something like

`#define SDFAT_USENAMESPACE` and then it would throw the whole thing inside of `sdfat::` - that way you could use this next to the audio system without any errors.

FrankBoesing, 2020-04-30:
Open an Issue here:
https://github.com/PaulStoffregen/Audio

As this is a long existing problem, it makes sense to add a reminder for Paul.

cfeied, 2020-08-30:
It has been a few months and the OP must surely have resolved this problem by now.  I just want to say that I am running SdFat-Beta with the audio library without difficulty.  To get rid of a large number of compilation errors I had to make only a very small number of changes in the audio library.  Fewer than 10 lines total IIRC.  Somebody else has made altered files available (slightly different from mine because I've made other patches over time) -- I'll attach them here.

I'm using SdFat-Beta for exFat cards on the built-in teensy 3.6 SD adaptor using FIFO_SDIO along with Fat32 cards on the AudioBoard SD adaptor using SPI_SHARED. I'm also using serial flash memory on the Teensy AudioAdaptor using the same SPI bus. I can read, write, copy, and move files among these three drives without difficulty. I play sounds from the SerialFlash chip or from SD02 while simultaneously writing sensor data to SD01, and it all works perfectly.

I'm attaching the changed files for the audio library.

[AudioLibChanges.zip](https://github.com/greiman/SdFat-beta/files/5146371/AudioLibChanges.zip)

PaulStoffregen, 2021-02-06:
Teensyduino 1.54 (beta) has completely removed the old Arduino SD library and now uses SdFat through a thin compatibility wrapper.

Hopefully that will fully solve this problem.