greiman/SdFat-beta #46: esp32 exFAT SD not initialising

Podgyhodgy, 2020-05-15:
I have an esp32 using hardware SPI with a tft display and the sd card sharing the SPI.

It is working fine with a FAT32 sd card - the SPI frequency is set to 16Mhz for the sd card and the display. It still works fine using the new library.

The problem arises when I use a new SanDisk Extreme Pro SD card (64Gb).
I have got as far as begin() working but when it tries to open a file it fails.

I have run bench.ino and it fails at the same point (opening the file) though it ran quite happily in the same hardware setup with a fat32 card.   The card is freshly formatted using the proper formatter.

 I have 
SD_FAT_TYPE 3
My chip select is fine on pin 5 for other cards.  SD_CS_PIN 5
I have SPI_CLOCK SD_SCK_MHZ(16)
I have forced the use of #define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)

 The output of bench is:-
Type is exFAT
Card size: 63.86 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SR64G
Version: 8.0
Serial number: 0X78524428
Manufacturing date: 1/2014

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
error: preAllocate failed
SdError: 0XC,0X7F

Thanks for your efforts.  
I am a bit new to mucking around with SD cards so forgive me if I am being foolish.
Glyn

greiman, 2020-05-15:
FreeStack not defined is OK.  It is not supported for ESP. On other systems it returns unused stack space.

I can't help with your problem.  I don't use ESP32 any more.

Podgyhodgy, 2020-05-15:
I am quite happy to do some investigation myself.  I am quite well equipped electronically (if need be) and I am good enough at software to have altered the Utouch library to work with esp32.  Could you perhaps briefly explain the preAllocate failed error or point me at some documentation?
 At the moment I do not know where to start to approach this error.

greiman, 2020-05-16:
Your error code is for a a multi block read which makes little sense. I can't help. 

You could try editing SdFatConfig.h and select standard library driver like this:

`#define SPI_DRIVER_SELECT 1`

It will be slower than the custom driver.  I will probably drop the custom driver since it is a pain to maintain.

I do have an ESP-VROOM-32 board and tried it with a 64 GB Samsung PRO+ SD.

I selected 25 MHz SPI speed since the ESP32 driver has a bug and tries speeds higher than it can support.

`#define SD_FAT_TYPE 3`

Here is the bench result:
```

Use a freshly formatted SD for best performance.

Type any character to start
Type is exFAT
Card size: 64.02 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X2C06AFA0
Manufacturing date: 9/2015

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2226.04,251,228,228
2225.05,251,228,228

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2130.24,248,238,239
2131.15,248,238,239
```

I tried shared SPI.  Here is the result.

```
Type any character to start
Type is exFAT
Card size: 64.02 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Version: 1.0
Serial number: 0X2C06AFA0
Manufacturing date: 9/2015

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
435.70,19854,284,1174
435.47,21840,284,1174

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
838.45,2708,471,609
839.44,1999,470,608
```

noorhaq, 2021-08-22:
Hi I wanna try exfat with esp32 to access some times like last modified last access etc. But last on some forums I checked esp32 doesn't support exfat so will it work

greiman, 2021-08-22:
There are conflicts with the esp32 system.  It's best to use the esp32 SD library.  I don't want to fight changes in esp32.

You can try SdFat but expect conflicts.