greiman/SdFat-beta #48: RTCTimeStampTest does not compile with ESP8266WiFi.h included

Primus007, 2020-05-18:
Using Arduino IDE.

The Sample RTCTimeStampTest (but some others also) does not compile with ESP8266WiFi.h included

#include <ESP8266WiFi.h>

if (!file.open("RtcTest.txt", FILE_WRITE)) {
exit status 1
'class fs::File' has no member named 'open'

How can i fix it?

greiman, 2020-05-19:
Looks like there is a conflict for the class File in esp8266 core.
```

C:\Users\bill\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.2\cores\esp8266/FS.h:52:7: note: previous declaration 'class fs::File'

 class File : public Stream
```

Guess you can't use SdFat with #include <ESP8266WiFi.h>.

Use the esp8266 SD.h package, it defines open for esp8266.

Looks like I need to drop support for ESP.

Primus007, 2020-05-19:
esp8266 SD.h does not support timestamp.
It is very difficult to find a good library that supports timestamp for files and runs with ESP8266WiFi.h

greiman, 2020-05-19:
The SdFat timestamp code in in esp8266 core but I don't see how a user can access it.

Here is the location of the version of SdFat used in ESP8266:

/esp8266/hardware/esp8266/2.7.1/libraries/ESP8266SdFat

Primus007, 2020-05-19:
it should work with: SdFile::dateTimeCallback(dateTime);
There is an example scetch "Timestamp"
I have not tested it.