greiman/SdFat-beta #51: System hangs when repeatedly triggering a wav file on Teensy 4.0

alexrjohnston, 2020-06-18:
Hello! first off thanks for the great lib. its amazing how many projects must be using SDfat now : )

I am trying to rid my system of clicks and pops when loading multiple WAVs on Teensy 4.0, several people have mentioned the issues with SD.h so I am trying sdfat on T4. SdFormatter, SdInfo and TeensySdioDemo are all working well! I can even get one wav playing using my own code and its seems the dropouts when a single wav is triggered repeatedly are much improved. Sadly when I try and trigger 4 wavs at the same time, although they play I loose control of the system at the serial window until they have finished playing. If I trigger two wavs at the same time, I can retrigger a couple of times and then I loose control. I also have the cash register sound that I can trigger as it is handy for debug.

Here is the code... (super lean to keep it simple)

`#define USE_SdFat_ // uncomment for SdFat.h, comment for SD.h lib.
#include "SdFat.h"
SdFat sd;

#include <Audio.h>
#include <Wire.h>
#include <SPI.h>

// GUItool: begin automatically generated code
AudioPlayMemory          sound0;         //xy=257,200
AudioPlaySdWav           playSdWav2;     //xy=296,375
AudioPlaySdWav           playSdWav3;     //xy=313,439
AudioPlaySdWav           playSdWav1;     //xy=324,292
AudioPlaySdWav           playSdWav4;     //xy=345,516
AudioMixer4              mixer1;         //xy=802,203
AudioMixer4              mixer2;         //xy=851,322
AudioOutputI2S           i2s1;           //xy=1069,248
AudioConnection          patchCord1(sound0, 0, mixer1, 1);
AudioConnection          patchCord2(sound0, 0, mixer2, 2);
AudioConnection          patchCord3(playSdWav1, 0, mixer1, 0);
AudioConnection          patchCord4(playSdWav1, 1, mixer2, 1);
AudioConnection          patchCord5(mixer1, 0, i2s1, 0);
AudioConnection          patchCord6(mixer2, 0, i2s1, 1);
// GUItool: end automatically generated code

// Use these with the Teensy Audio Shield
//#define SDCARD_CS_PIN    10
//#define SDCARD_MOSI_PIN  7
//#define SDCARD_SCK_PIN   14

// SDCARD_SS_PIN is defined for the built-in SD on some boards.
#ifndef SDCARD_SS_PIN
const uint8_t SD_CS_PIN = SS;
#else  // SDCARD_SS_PIN
// Assume built-in SD is used.
const uint8_t SD_CS_PIN = SDCARD_SS_PIN;
#endif  // SDCARD_SS_PIN

// Use these for the SD+Wiz820 or other adaptors
//#define SDCARD_CS_PIN    4
//#define SDCARD_MOSI_PIN  11
//#define SDCARD_SCK_PIN   13
int inByte = 0;
byte  newDATAin = 1;

#include "AudioSampleCashregister.h" // http://www.freesound.org/people/kiddpark/sounds/201159/

void setup() {
  Serial.begin(115200);

  // Audio connections require memory to work.  For more
  // detailed information, see the MemoryAndCpuUsage example
  AudioMemory(100);

  //SPI.setMOSI(SDCARD_MOSI_PIN);
  ///SPI.setSCK(SDCARD_SCK_PIN);
  if (!sd.begin(SdioConfig(FIFO_SDIO))) {
    // stop here, but print a message repetitively
    while (1) {
      Serial.println("Unable to access the SD card");
      delay(500);
    }
  }   
}

void loop() {
   
  
   
switch (Serial.read()) {

            
      
   case 's': {
    Serial.println("s");
   // envelope1.noteOff();
   // envelope2.noteOff();tttttttttttt
   playSdWav1.stop();
   playSdWav2.stop();
   playSdWav3.stop();
   playSdWav4.stop();
   //playSdWav5.stop();
   //playSdWav6.stop();
   //playSdWav7.stop();
   //playSdWav8.stop();
   sound0.play(AudioSampleCashregister);

 
   
 }
 break;

 case 't': {

  sound0.stop();

  playSdWav1.play("10BAB1.WAV");

  playSdWav2.play("10BAB2.WAV");
  
  playSdWav3.play("10BAB3.WAV");
  playSdWav4.play("10BAB4.WAV"); 

 }
 break;

 case 'y': {

  sound0.stop();

  playSdWav1.play("10BAB2.WAV");

 delay(5);

  //playSdWav2.play("10BAB2.WAV");
  
  //playSdWav3.play("10BAB3.WAV");
  //playSdWav4.play("10BAB4.WAV"); 

 }
 break;

   
}
}`

Here is the audio used in the above example: http://donjohnstonmusic.co.uk/tr/sinewavs.zip

I am aware three of the wav players are not connected to any outputs, they still play and cause the issue and its the same when they are connected. is there something in the SDfat settings I can tweak to prevent this hang?

all help very gratefully received, I am properly stuck

greiman, 2020-06-18:
SdFat uses two 512 byte cache buffers.  One for FAT entries and one for file data.  If you open multiple files, you will have problems since these caches will be invalidated often.

SdFat evolved from a 2008 library for an Arduino Diecimila with a ATmega168, 1 KB SRAM, and 16 KB flash and a single 512 byte sector cache.

It would be possible to extend SdFat for provide cache for each file but I have lost interest in developing SdFat.

SD cards are not intended to be multi stream devices without huge buffers and file read-ahead.

Teensy 4.0 is a nice board but is lacks a decent multi-tasking OS with an integrated file system that can cope with multiple data streams.

alexrjohnston, 2020-06-20:
Thanks for taking the time to respond! Totally understandable, that all makes good sense. Do you think a pi running a light as possible os would be better suited for multiple data steams?

greiman, 2020-06-20:
Pi has a better OS since the base is Linux.  It depends on how the file system cache is configured.

alexrjohnston, 2020-06-20:
Thanks for the replies I will look into it! Regards, Al.

PaulStoffregen, 2021-02-06:
Hopefully this has fixed the problem.

https://github.com/PaulStoffregen/Audio/commit/6369a6ae1f3b6ea547a5192f9bf7efe3b778a629