greiman/SdFat-beta #52: LGT8Fx support

Oxmaster, 2020-07-20:
I'm checking out the ATmega 328 clones by Logic Green (LGT8Fx). There is _some_ support as these chips are not well known (https://github.com/dbuezas/lgt8fx). I have the 328P-SSOP20 (pro mini clone) and 328P-LQFP32 (nano clone)
Beforehand I tested the SD card and the library with Arduino Nano clone (328P-MU QFN32 with CH340G) and everything works. I'd like to help to make this library support the LGT8Fx chips but I don't know too much about this low level stuff that it implies, but can do some tests and relay the results.
Here is SdInfo output from 328P-MU Nano at 16 MHz, the card was formated via the SdFormatter sketch.
```SdFat version: 2.0.0

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 9
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 21 ms

Card type: SD2

Manufacturer ID: 0X2
OEM ID: TM
Product: SA01G
Version: 0.1
Serial number: 0X6FBD009C
Manufacturing date: 5/2008

cardSize: 1019.22 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0X80FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X1,0X1,0X0,0X6,0XFE,0XFF,0XFF,249,1990407
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT16
sectorsPerCluster: 32
clusterCount:      62184
freeClusterCount:  62184
fatStartSector:    250
dataStartSector:   768
```
At 32 MHz from SdInfo sketch: 
```
init time: 166 ms
readInfo failed
SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x18
SD errorData = 0x1
```

And at 16 MHz (same for 8 MHz):
```
init time: 4 ms

Card type: SD2

Manufacturer ID: 0XFE
OEM ID: T
Product: S0MG
Version: 4.1
Serial number: 0X1BD319C
Manufacturing date: 0/2000

m_csd version error

OCR: 0X80FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X1,0X0,0X0,0X0,0XFE,0X1,0XFF,16711686,24569
2,0X7,0X0,0X1E,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

MBR not valid, assuming Super Floppy format.

volumeBegin failed. Is the card formatted?
```
There is something strange with OEM ID and Product.

Also tested SoftwareSPI sketch and it works at any clock speed.

I'm unsure if it's hardware or software issue.

greiman, 2020-07-20:
I don't have this hardware so you must solve this problem.

closed 2020-07-20 by greiman

greiman, 2023-01-27:
I am no longer adding custom drivers.  There are now hundreds of "Arduino like boards".

SdFat version 2.x probably will fail with a compile error for  LGT8F328P if USE_STANDARD_SPI_LIBRARY  is zero.