greiman/SdFat-beta #55: Possible bug: DBG_FAIL triggered when dir.exists(filename) is called and filename does not exist

cfeied, 2020-08-30:
when a file does not exist, checking for file existence returns false as expected. However, if I enable debugging it also triggers an unexpected DBG_FAIL before returning:

`DBG_FAIL: FatFileLFN.cpp 418`

the relevant code is:
```
create:
  // don't create unless O_CREAT and write mode
  if (!(oflag & O_CREAT) || !isWriteMode(oflag)) {
    DBG_FAIL_MACRO;
    goto fail;
  }
```
Although everything works fine with debugging turned off, this still seems like a possible bug because I can't see any reason why the code should end up here after an attempted open in read mode. My environment won't let me see a call stack so I can't easily backtrack.

greiman, 2020-08-30:
I never intended for debug to be use for finding bugs in applications.  It is intended to find bugs in the library.  I try to place a debug macro everywhere a fail is returned and exists will return false. 

The problem is that open will fail if you attempt to open a file that does not exist for read or write without O_CREATE.

There are many other quirks if you use debug to find bugs in applications.

cfeied, 2020-09-04:
Yes, I was looking for bugs in the library because some of the results were not as expected. However, it turned out that the results were as I *should* have expected, had I understood the library better.

It just seemed odd that the library fires a debug_fail on the *expected* outcome of a function (testing for file existence returns false when the file is not present).  It made me scrutinize the library code when in fact it was performing as designed.

closed 2020-09-04 by cfeied