greiman/SdFat-beta #58: Question about ExFatLogger

jerabaul29, 2020-10-07:
Hi again,

I thought a bit after our discussion here  https://github.com/greiman/SdFat/issues/210 I think I will give a try to your beta actually :)

I am looking at your ExFatLogger

https://github.com/greiman/SdFat-beta/blob/master/examples/ExFatLogger/ExFatLogger.ino

It looks like it writes data as 512 bytes block always, right?

https://github.com/greiman/SdFat-beta/blob/8adf5b5a86bab72c68a4fd6b32f45c4366c05209/examples/ExFatLogger/ExFatLogger.ino#L350

Does that mean that even on modern cards, I should write in chunks of 512 bytes always when using the code for getting best performance? :) Do you know if there are some blog posts for newbies like me to learn more about these SD card subtleties?

jerabaul29, 2020-10-07:
And one more thing: in this case, it looks like we do not always write chunks of always exactly 512 bytes, right? If I can manage to always organize my data in chunks of size exactly 512 bytes, may it help to boost speed even more?

greiman, 2020-10-07:
The only option for an SD is to transfer in units of 512 bytes.  Think of 512 as a packet size, it has little to do with how the card programs flash.  Often a card copies an entire 512 KB flash page when you modify a block.

Cards are designed for devices like phones with GB of memory.  Cards are designed for SDIO, not SPI.

You should forget trying to optimize, cards have various internal buffers and you can only find out about the internals of a card if you have a nondisclosure agreement with the manufacture.

jerabaul29, 2020-10-07:
Ok, thanks !

Regarding this library Vs the old one: what is the difference then if still 512 bytes are written at a time? Is it purely the SPI implementation + some internal stuff? Anything to do with how clusters are handled? Just being curious.

I think when I have time that I will try to write some form of blog post about my modest user experience. I will let you know. But I will try to make sure what you explain here gets propagated further :)

greiman, 2020-10-08:
I am closing this I can't simply explain how an SD library works.

closed 2020-10-08 by greiman