greiman/SdFat-beta #59: hosting the documentation on a github.io page

jerabaul29, 2020-10-07:
Do you think it would be possible to host your html files on a github.io page? It would make it much easier to browse :)

greiman, 2020-10-07:
No, I find almost no one reads the documentation.  I wrote the first version of SdFat over ten years ago and a version was used in the official SD.h library.  

Most people just use the simplest open, print, close features.  I have tried to optimize SdFat so you don't need to know more and may even reduce performance by trying to optimize.

jerabaul29, 2020-10-07:
Ok, thanks. So you actually advise to use the latest library, and not care myself about performance, but let the library take care of it, right? :)

closed 2020-10-08 by greiman