greiman/SdFat-beta #60: SD card error with beta version

jvoermans, 2020-10-22:
Hello,

I'm trying to write sensor data to an SD card but am running into some problems.
I'm using an Arduino Due and Adafruit SD Ultimate Breakout. It works all fine when using the old SdFat library, for example, the LowLatencyLogger.

However, when using the ExFatLogger (SdFat-Beta library) with exactly the same wiring, I am getting the following error:

begin() failed
Do not reformat the SD.
SdError: 0X18,0X0

Any thoughts?

My SD-Card has the following details:

init time: 5 ms
Card type: SDHC

Manufacturer ID: 0X3
OEM ID: SD
Product: SC16G
Version: 8.0
Serial number: 0X868857A2
Manufacturing date: 12/2012

cardSize: 15931.54 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,31108096
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is FAT32
sectorsPerCluster: 16
clusterCount:      1942208
freeClusterCount:  1942202
fatStartSector:    10612
dataStartSector:   40960

jerabaul29, 2020-10-22:
(how did you get these details / by running which script @jvoermans ? :) ).

jvoermans, 2020-10-22:
Sorry, forgot to mention. SD card details are from the SdInfo example sketch

jerabaul29, 2020-10-22:
Ok, so you are able to run SdInfo from SdFat-beta, but not ExFatLogger from SdFat-beta, right? Strange.

jerabaul29, 2020-10-22:
PS: I can run both scripts with Arduino due and the same breakout board module from Adafruit, but using a different SD card.

greiman, 2020-10-22:
The problem may be that the combination of your card and the Adafruit breakout will not run at 42 MHz on the Due with SdFat-beta for the ExFatLogger.  The SdInfo example limits the SPI rate to 16 MHz.

The Due does not require level shifters and I have found using level shifters on Due can be marginal for some applications with both SdFat and SdFat-beta.  The level shifters distort the SPI signal at max speed.

I use a breakout designed for 3.3V on Due.  I search for "SD Card Module For Arduino ARM" on ebay.

Try this which will limit the speed to 21 MHz.
```
// Select the fastest interface. Assumes no other SPI devices.
#if ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(25))
#else  // ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(25))
#endif  // ENABLE_DEDICATED_SPI
```

jerabaul29, 2020-10-22:
i have been using the same hardware (Due + Adafruit SD breakout) but with a different SD card, and it works fine with all the codes. So I think this may be related to the card as you say.

jvoermans, 2020-10-22:
@greiman Great, works like a charm!
Thanks a lot :)