greiman/SdFat-beta #69: Support for Multiple Partitions

mjs513, 2021-02-21:
Morning Bill
Have a quick question.  I have a sd card (128GB) that has 2 partitions on it and it looks like the library has some functions to support multiple partitions but not sure how to use them.   Plus have seen them up to 1TB, unbelievable.

Guess first - does SDFat-beta support multiple partitions.

What I was looking at was that if you do a 
`FatPartition Part1`
Sets up and instance for a fat partition but I am kind of stuck on init
`Part1.init(Blockdevice* dev, part)`
not sure what Blockdevice * dev points to.  I think its the SDcard but not sure how to identify that - doesn't look I can get access to that unless its in the card() structure but I don't know where to find that to check

greiman, 2021-02-21:
SD cards are not designed for multiple partitions and the SD association specifies that a card shall have only one partition.  The structure of a card's flash, buffers, and algorithms depends on not just one partition but the location of file system structures and cluster sizes.

Here is the quote from the file system spec: 

> This specification specifies some file systems. The type of the file system to be used shall be uniquely decided with Card Capacity as follows. Here, Card Capacity means the total size of Data Area and Protected Area size.  
> 
> File System Type That is, all Standard Capacity SD Memory Cards whose capacity is 2048MB or less shall use FAT12 / FAT16 file system, and never use the other file system. Similarly, all High Capacity SD Memory Cards shall use FAT32 file system, and never use the other file system. And all Extended Capacity SD Memory Cards shall use exFAT file system, and never use the other file system. This includes the prohibition of partial format of SD Memory Card. For example, 8GB High Capacity SD Memory Card should not be formatted as 2GB card with FAT12 / FAT16 file system. In this case, whole area of 8GB should be formatted with FAT32 file system.

Here is the MBR spec:

> (BP 446 to 461) Partition Table (partition1)
> This field shall specify the information of first partition in the volume. This partition means Regular Area
> that user can access without mutual authentication. It shall be recorded according to Table 5-2.
> 
> (BP 462 to 477) Partition Table (partition2)
> This field shall be recorded as ZEROs, as a volume shall consist of single Regular Area.
> 
> (BP 478 to 493) Partition Table (partition3)
> This field shall be recorded as ZEROs, as a volume shall consist of single Regular Area.
> 
> (BP 494 to 509) Partition Table (partition4)
> This field shall be recorded as ZEROs, as a volume shall consist of single Regular Area.
> 
> (BP 510 and 511) Signature Word
> This field shall be recorded as 55h (BP 510) and AAh (BP 511).

I wrote the FatLib and ExFatLib to support the FAT and exFAT spec but limited SD cards to the official SD spec.

mjs513, 2021-02-21:
Thanks for the detailed explanation, at least it makes sense now.  Did have a feeling you were going to tell me that.  Was also looking at it support SSD's via USBHost but guess have to do something else.

BTW it does read the SSD 1st partition without a problem :).  Tried it :)

Thanks again
Mike

closed 2021-02-21 by mjs513