greiman/SdFat-beta #71: Maybe support __has_include for configuration for sketch or library override of configuration.

KurtE, 2021-03-05:
Wondering your thoughts of allowing sketches and/or libraries a way to override the settings in the SdFatConfig.h file.

For example the SdFatConfig.h file near the end (probably before last endif

could have something like:
```
#if defined __has_include
#  if __has_include (<UserSdFatConfig.h>)
#    include <UserSdFatConfig.h>
#  endif
#endif
```

Which maybe the users (or library) could for example set:
```
#define MAINTAIN_FREE_CLUSTER_COUNT 1
```
Actually sort of wondering why this is not on by default, at least on arm

greiman, 2021-03-05:
I have lots of requests like this.  Some for platformIO, some for make, and other IDE environments.   I have not decided on a design that would satisfy most of these requests.

I know I can't just allow a replacement config file with a user file since config files change when new features are added.  A user include file could override selected symbols.

I use a lot of RTOS systems that have the config problem.  Most RTOS systems assume the user can easily define individual symbols like in eclipse IDE then the default config file has something like this:
```
/**
 * @brief   System time counter resolution.
 * @note    Allowed values are 16, 32 or 64 bits.
 */
#if !defined(CH_CFG_ST_RESOLUTION)
#define CH_CFG_ST_RESOLUTION                32
#endif

/**
 * @brief   System tick frequency.
 * @details Frequency of the system timer that drives the system ticks. This
 *          setting also defines the system tick time unit.
 */
#if !defined(CH_CFG_ST_FREQUENCY)
#define CH_CFG_ST_FREQUENCY                 10000
#endif
```

So it's on my too do list but I will not be adding any features to SdFat for some time.  I have started a major restructuring of SdFat for a number of new features and changes.

KurtE, 2021-03-09:
Sounds good. 

Yes I have seen options like you mentioned, where you only define it if it is not already defined.  This does allow systems to do it in different ways, like, change their Arduino build (probably platform.txt) to add in a -DCH_CONFIG_ST_TIMEDELTA=4
or something close to that.  

Or as I mentioned, some have gone to using the _z-has_include which I have seen some libraries, Like the teensyTimerTool library has 

```
#pragma once

#if __has_include("userConfig.h")
    #include "userConfig.h"
#else
    #include "defaultConfig.h"
#endif
```

Which is probably too generic of names... I have seen and done others, that for example either put in the test at the start of the main options header or at the end... So for example at the start, could allow them to set some of them, which then if you have those with the syntax you mentioned, would allow that file to set those settings... 

However with Arduino there is a limitation.  Those files can not be in sketch directory.  I did have luck with one library being able to define it to help control a different library. 

I totally understand that you are busy and won't get to any of this for a long time and maybe never.  But thought I would throw out the idea.

Also not sure of best place to maybe discuss some things like, how to speed up getting the count of free clusters on a Fat32 system.  Especially with MSC work going on to allow things like thumb drives to work that are plugged into USBHost. 

The current checked in code on a few GB size can be real slow on very large partitions.  As you know the code reads in one sector, scans that sector reads in the next sector...  And each read sends out a USB SCSI like command to read one sector.  So one write out plus wait to receive...

I wrote up another specific function in test sketch that uses some of the USBHost code, where instead I ask it to read in all of the sectors of the fat, and call a callback function when each sector is received.    My quick tests showed this was about 10-12 times faster.  

Then wwatson mentioned the FS Information block which keeps a cached count...  So wrote up quick and dirty version of code to read that, which again is faster... Could be faster still if the FatPartition begin code kept information... 
A sample run run today by wwatson of this code shows: 
```
1:>> Fat32: Volume name:(32GFAT32) Partition Total Size:14027751424 Used:32776192 time us: 5068874
    Free Clusters: API: 1708371 by CB:1708371 time us: 836490
    Free Clusters: Info: 1708371 time us: 1492
32MEGfile.dat
```
Question is does it make sense for some integration of this stuff into your library?  Or at least make it easier for external to integrate.  For example, with the MTP/MSC integration, I could see wanting the ability to read in the info sector and have your option: MAINTAIN_FREE_CLUSTER_COUNT 
allow the function to set the free cluster count from this sector and then if MTP writes or the like happen, like create a new file or delete a file.  When I complete the operation, maybe in addition to telling MTP that the data needs to be updated for storage, I might write out an update FSInfo sector...  Would probably also want a way to say, this data is no longer valid and then do a scan of the data...

Again not sure if you are interested in such things?  Best to discuss on github?  Or 
for example there are a few different threads up on PJRC like: 
https://forum.pjrc.com/threads/66338-Many-TLAs-MTP-MSC-FS-SD-SDFat-LittleFS-UsbMSCFat-to-work-with-each-other-8)?p=272825#post272825

Again I understand you are busy, but thought I would mention some of the stuff we are playing with.

Good luck with the ChibiOS

closed 2021-03-09 by KurtE

greiman, 2021-04-03:
I made most SdFatConfig.h defines conditional so the symbols can be defined in a makefile or an IDE like PlatformIO.

Also UTF-8 encoded Unicode filenames for both FAT and exFAT.

KurtE, 2021-04-03:
Thanks, sounds good.