greiman/SdFat-beta #72: Warning "'next' may be used uninitialized in this function"

FrankBoesing, 2021-06-29:
[https://github.com/greiman/SdFat-beta/blob/790dd84e26e2b7a7bd91d3ee355a4b2855694db5/src/ExFatLib/ExFatPartition.cpp#L225](https://github.com/greiman/SdFat-beta/blob/790dd84e26e2b7a7bd91d3ee355a4b2855694db5/src/ExFatLib/ExFatPartition.cpp#L225)

GCC 10 prints the mentioned warning.

greiman, 2021-06-29:
next is always set at line 225 by this.
```
    status = fatGet(cluster, &next);
    if (status < 0) {
      DBG_FAIL_MACRO;
      goto fail;
    }
```
 I will fix the warning by initializing next to zero.

closed 2021-06-29 by greiman

greiman, 2021-06-29:
It appears the program works OK but I made two changes that may suppress the warning.

First the return type of fatGet is wrong.  it should be int8_t but all calls use int8_t for tests.

next is not used unless the return status is one.  I changed this test to not use next unless status is one.

`if (status == 0 || (cluster + 1) != next) {`

I should probably change the loop to not execute these assigns before returning.

```
      start = next;
    }
    cluster = next;
  } while (status);
```

FrankBoesing, 2021-06-29:
Thank you, Bill !