greiman/SdFat-beta #74: cannot open File with UTF-8 Name

SWIPHER, 2021-08-02:
Hi,
I'm trying to open a chinese name file like "你好.txt" but it doesn't work;
and the major code is like:
`#define USE_UTF8_LONG_NAMES 1 /*And I also change the value of this in SdFatConfig.h*/
SdFs sd;
FsFile music;
sd.begin(SD_CS);
music.open("你好.txt");`
Looking forward for reply
Thank you so much!

greiman, 2021-08-02:
I have no idea what your problem is.

I added your filename to the UnicodeFilenames example like this.
`const char* names[] = {u8"你好.txt", u8"россиянин", u8"très élégant", u8"狗.txt", nullptr};`

I set the following in the example and ran it on FAT16, FAT32, and exFAT.
```
// Remove files if non-zero.
#define REMOVE_UTF8_FILES 1

// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 3
```
The output is:

Type any character to begin
```
ls:
      131072 😀/
            12 你好.txt
            20 россиянин
            17 très élégant
             9 狗.txt
Done!
```
I can open the files on a PC and they have the correct content.

SWIPHER, 2021-08-03:
Hi,I solved it just now.
I found the reason is In Windows Chinese filenames always encode with GBK not the UTF-8
So I chang The System language to UTF-8 encode and rename the file to UTF-8 format.
Now it works fine.
Thank you so much