greiman/SdFat-beta #76: ESP32 check sd format

SyXavier, 2021-09-24:
Hi, Mr greiman. I configured SdFatConfig.h, but the serial port returns check sd format. The test case I used is unicodeFilename.

greiman, 2021-09-24:
How is the SD formatted and how did you format it?

What is the output from the [SdInfo](https://github.com/greiman/SdFat-beta/tree/master/examples/SdInfo) example?

SyXavier, 2021-09-24:
I use the windows disk tool to format the sd card to FAT32. The sd card can be read and written normally on the PC side.

---------------------------------------------------------------------------
SdFat version: 2.1.0-beta.3

Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 5
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 331 ms

Card type: SDHC

Manufacturer ID: 0X73
OEM ID: BG
Product: NCard
Version: 1.0
Serial number: 0XAB577019
Manufacturing date: 10/2011

cardSize: 16021.19 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X73,0X73,0X20,0X61,0X6E,0X79,0X20,0X6B,1948285285,1701978223
2,0X73,0X74,0X61,0X72,0X74,0XD,0XA,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,28049408,441

MBR not valid, assuming Super Floppy format.

volumeBegin failed. Is the card formatted?

type any character to start

greiman, 2021-09-24:
This card has a non-standard format.  Try the [SD formatter](https://www.sdcard.org/downloads/formatter/).

The card is very strange, 1.6 GB.

SyXavier, 2021-09-25:
Thank you for your help, the problem is solved. I am grateful that you created this project to support unicode encoding.

uzi18, 2021-09-25:
@SyXavier please try to explain how did you solve it for others in the future

SyXavier, 2021-09-25:
@uzi18 Use SD Formatter to format the SD card.