greiman/SdFat-beta #78: 'SdFat {aka class SdFs}' has no member named 'vwd'

plumped, 2021-11-20:
Hey Guys,

Im trying to run the GameFrame Code from https://github.com/Jerware/GameFrameV2 with a Teensy 4.1

My Code: 
`#include <SPI.h>
#include <EEPROM.h>
#include <Wire.h>
#include <Time.h>
#include <DS1307RTC.h>
#include <SdFatBeta.h>
#include <IniFileLite.h>
#include <FastLED.h>
#include <Entropy.h>
#include <IRremote.h>

/***************************************************
  Game Frame V2 Source Code
  Jeremy Williams, 2-26-2016

  Game Frame is available at LEDSEQ.COM

  NOTE: Altering your firmware voids your warranty. Have fun.

  BMP parsing code based on example sketch for the Adafruit
  1.8" SPI display library by Adafruit.
  http://www.adafruit.com/products/358

  In the SD card, place 24 or 32 bit color BMP files
  (Alpha channel is ignored in 32-bit files.)

  There are examples included
 ****************************************************/

#define SD_CS    10  // Chip select line for SD card
#define SD_FAT_TYPE 2
SdFat sd; // set filesystem
SdFile myFile; // set filesystem

#define BUFFPIXEL 16 // number of pixels to buffer when reading BMP files

// Time
tmElements_t tm;

// IR setup
const byte RECV_PIN = 4;
IRrecv irrecv(RECV_PIN);
decode_results results;
char irCommand;
char irLastCommand;
boolean understood = false;

// LED setup
#define DATA_PIN 2
#define CLOCK_PIN 3
#define NUM_LEDS    256
#define LED_TYPE    SK9822
#define COLOR_ORDER BGR
CRGB leds[NUM_LEDS];

//Button setup
const uint8_t buttonNextPin = 16; // "Next" button
const uint8_t buttonMenuPin = 17;  // "Menu" button

#define STATUS_LED 6

//Enable verbose prints?
const boolean debugMode = false;

//Global variables
[... 3073 lines cut ...]
  return result;
}

uint32_t read32(SdFile& f) {
  uint32_t result;
  f.read(&result, 4);
  return result;
}

// clock debug function
void print2digits(int number) {
  if (number >= 0 && number < 10) {
    Serial.write('0');
  }
  Serial.print(number);
}`

Any idea why the SdFat always yelling at me that there is no 'vwd' member?
The code was written in the past for compiling with teensy 3.x 

The connected SD card doesn't matter. I have tried different cards from different manufacturers and types & sizes.

The files are written to root -> "\" extracted from the files from https://github.com/Jerware/GameFrameV2/tree/master/sd

Thank you!

greiman, 2021-11-20:
There is no vwd() in SdFat V2.x. for SdFs since I could not find a reliable way to support  vwd for FAT16/FAT32 and exFAT for types SdFat, SdExFat and SdFs.  There is a private vwd() for classes SdFat and SdExFat that is not user accessible.

Also don't want a user accessible vwd() since it has caused  problems because of conflicts between internal use and application use.

For users up to now it has been easy to simply open a directory file when using open by index.

```
 FsFile cwd; 
 cwd.open(dirPath);
 open(&cwd, index, mode);
```

Looks like I need a easy way to open a file in the current volume working directory.

Maybe something like file.open(&sd, index, mode).  I already have a file.open(&sd, path, mode) it is used in open(path, mode):
```
bool ExFatFile::open(const char* path, oflag_t oflag) {
  return open(ExFatVolume::cwv(), path, oflag);
}
```

Then I could also have an file.open(index, mode) that opened a file by index on the current working directory.

It will be simple to do but will be a while since I have lots of testing for other features in the next version.

plumped, 2021-11-20:
Thanks for your quick feedback!
So is there any workarround known for the requested? Or is the easiest way to buy a teensy 3.x? xD

Im a complete noob with teensy btw.

greiman, 2021-11-20:
No easy workaround and you will have the problem with any V 2.x of SdFat.  I guess Teensy 3.x still supports V 1.x.

A not so easy workaround is to add a cwd file:
`SdBaseFile cwd;`

Add this function
```
void sdChdir(const char* path) {
 cwd.close();
 cwd.open(path);
 sd.chdir(path);
}
```
and replace sd.chdir(path) with sdChdir(path) where:

replace sd.vwd() with &cwd like this:

` myFile.open(sd.vwd(), fileIndex, O_READ);`

becomes 

` myFile.open(&cwd, fileIndex, O_READ);`

plumped, 2021-11-21:
thank you for the explanations! I'll try the whole thing times yet, but have also ordered a 3.6 for current use. You are now on my watchlist and I look forward to the new features. 

For 4.x is still the beta variant is that correct so far?

Have a nice sunday!

greiman, 2021-11-21:
> For 4.x is still the beta variant is that correct so far?

I don't know which version is used in Teensy.  I don't develop the Teensy wrapper.

greiman, 2021-11-25:
I added a call in SdFat-beta that opens a file by index in the current working directory which is volume working directory, sd.vwd(), if you have only one SD.

That should only require replacing: 
`myFile.open(sd.vwd(), fileIndex, O_READ);`
by
`myFile.open( fileIndex, O_READ);`

plumped, 2021-11-25:
Brilliant! Thank you!