greiman/SdFat-beta #79: SDIO multi-sector read timings

robertsonics, 2022-05-21:
I have an application running on a Teensy 4.1 which requires direct multi-sector reads from the SD card - I don't need FAT support. I achieved this with the SdFat-beta library by simply adding the following function to SdFat.h:

  bool readSectors(uint32_t sector, uint8_t * buf, size_t numSectors) {
      return m_card->readSectors(sector, buf, numSectors);
  }

I call this function to read 8 sectors (4K) at a time, and I've bracketed the call with a digital output that I can see on my oscilloscope so I can tell how long the operation takes.

On an Atmel SAMS70 with the SDIO bus running at 50MHz, this same operation takes about 1ms with a modern microSD card. On the Teensy 4.1 it consistently takes 2.7ms. If I instead read 16 sectors at a time on the Teensy, the operation takes just 3ms, and if I read 32 sectors at a time, it takes only 3.3ms. I know the reads are all good, because I'm confirming the data. Very infrequently, I see the 32 sector read operation take only about 1.5ms, again with no errors.

It appears as though there's a chunk of unexplained time, like around 2ms, involving a read of any size. I don't believe it's the microSD card because I don't see this using the same card with my SAMS70. I started looking at the SdioTeensy files and am wondering if there's a possibility that there's some minimum time spent waiting for the DMA operation that's longer than necessary?

Thought I'd ask here before diving in further.

robertsonics, 2022-05-21:
Sorry - that was premature. Turns out I wasn't using the microSD card I thought I was using. With the correct card (an SP 32GB "3D NAND" card) I'm now getting about 400us for an 8sector read operation and 550us for a 16 sector read. That's more like it!

greiman, 2022-05-21:
16 sectors in 550us is pretty good.  

You can't predict how cards will perform on a mpu like Teensy.  They are designed for huge transfers in a high speed  UHS mode with 1.8V siginaling.

closed 2022-05-21 by greiman