greiman/SdFat-beta #81: Some SDCard Samples work others not ,SdFat Lib

muekno, 2023-01-28:
Arduino IDE on MAC
Bord ESP8266 generic (well working with other sketches
Chipselect pin 15 = D8
SDFAT Library 2.2.1 Adafruit fork
Sdinfo.ino works fine that means the data about my 16GB SD high speed looks correct
DirectoryFunctions not see errors before
begin() failed
Do not reformat the SD
SdError: 0X19,0X0 - what means SD_CARD_ERROR_READ_TOKEN - Bad read data token, but this means nothing to me
the following 0xo means no error what I do not understand too

sdinfo.inoshows
19:07:45.474 -> init time: 3 ms
19:07:45.474 -> 
19:07:45.474 -> Card type: SDHC
19:07:45.474 -> sdSpecVer: 3.00
19:07:45.474 -> HighSpeedMode: true
19:07:45.474 -> 
19:07:45.474 -> Manufacturer ID: 0X2
19:07:45.474 -> OEM ID: TM
19:07:45.474 -> Product: SA16G
19:07:45.474 -> Revision: 2.5
19:07:45.474 -> Serial number: 0X263131CD
19:07:45.474 -> Manufacturing date: 3/2016
19:07:45.474 -> 
19:07:45.474 -> cardSize: 15489.56 MB (MB = 1,000,000 bytes)
19:07:45.474 -> flashEraseSize: 128 blocks
19:07:45.474 -> eraseSingleBlock: true
19:07:45.509 -> dataAfterErase: zeros
19:07:45.509 -> 
19:07:45.509 -> OCR: 0XC0FF8000
19:07:45.509 -> 
19:07:45.509 -> SD Partition Table
19:07:45.509 -> part,boot,bgnCHS[3],type,endCHS[3],start,length
19:07:45.509 -> 1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,30244864
19:07:45.509 -> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
19:07:45.509 -> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
19:07:45.509 -> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
19:07:45.509 -> 
19:07:45.509 -> Scanning FAT, please wait.
19:07:46.862 -> 
19:07:46.862 -> Volume is FAT32
19:07:46.862 -> sectorsPerCluster: 64
19:07:46.862 -> fatStartSector:    9000
19:07:46.862 -> dataStartSector:   16384
19:07:46.862 -> clusterCount:      472448
19:07:46.898 -> freeClusterCount:  472371

And SDFormatter shows
19:10:11.468 -> Warning, all data on the card will be erased.
19:10:11.468 -> Enter 'Y' to continue: Y
19:10:31.742 -> error: Get sector count failed.
19:10:31.742 -> SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x19
19:10:31.742 -> SD errorData = 0x20

The card was initialy formated with the free SD Card Formatter 

What I want is logging weather data on sd card bigger than 4GB, but this library seams to be the only one thatcan handel FAT23 or extFAT card

need some help, thamks
Rainer

greiman, 2023-01-28:
> SDFAT Library 2.2.1 Adafruit fork

Contact Adafruit for support of their fork.

greiman, 2023-01-28:
Your problem may be the flakiness of the ESP8266 hardware and the buggy SPI driver.

I request 50MHz SPI speed and the SPI driver should choose the highest supported speed.  Often ESP boards fail.

Try editing examples and reduce the requested SPI speed.  For example in the SdFormatter try 16 MHz.  I use 16 MHz in SdInfo.

 [Here](https://github.com/adafruit/SdFat/blob/5c0b73ada271658d64577738464a0ac2bf337a05/examples/SdFormatter/SdFormatter.ino#L40) is place to edit.

muekno, 2023-01-29:
Thank you,
I will try, the solution looks meaningfull.

Have a nice Sunday

Kind regards

Rainer 

>>> Bill Greiman ***@***.***> 21:32 28 Januar, 2023 >>> 
Your problem may be the flakiness of the ESP8266 hardware and the buggy SPI driver.

I request 50MHz SPI speed and the SPI driver should choose the highest supported speed.  Often ESP boards fail.

Try editing examples and reduce the requested SPI speed.  For example in the SdFormatter try 16 MHz.  I use 16 MHz in SdInfo.

 [Here](https://github.com/adafruit/SdFat/blob/5c0b73ada271658d64577738464a0ac2bf337a05/examples/SdFormatter/SdFormatter.ino#L40) is place to edit.