greiman/SdFat-beta #82: ESP32: SD-card on HSPI port

folkertvanheusden, 2024-02-10:
Hi,

I would like to connect my SD card reader to the HSPI port on an WEMOS32 as I also have an Ethernet board connected to the VSPI port.
So I connected SCK to 14, MISO on 12, MOSI on 13 and SS on 15.
This doesn't work unfortunately: `SdError: 0X17,0X7F`.
Is this supposed to work? If not: can I connect both SD & Ethernet to the "primary" SPI port? (with each their own SS of course)

```
#define SPI_DRIVER_SELECT 1
#include <SdFat.h>
#include <SPI.h>

#define MY_SS 15
SPIClass spi_sd(HSPI);
#define SPI_CLOCK SD_SCK_MHZ(16)
#define SD_CONFIG SdSpiConfig(MY_SS, SHARED_SPI, SPI_CLOCK, &spi_sd)

SdFs sd;

void setup() {
	Serial.begin(115200);

	Serial.println(F("Init SPI for SD reader"));
	spi_sd.begin(14, 12, 13, MY_SS);  // SCK, MISO, MOSI, SS

	if (sd.begin(SD_CONFIG) == false) {
		Serial.println(F("SD card init failed"));
		sd.errorHalt();
	}

	Serial.println(F("Go!"));
}

void loop() {
	sd.ls(LS_DATE | LS_SIZE);
	delay(1000);
}
```

greiman, 2024-02-10:
I can't help  since I have little experience with WEMOS32.

sfranzyshen, 2024-02-10:
> I can't help since I have little experience with WEMOS32.

why? WEMOS32 is just another ESP32 ...

greiman, 2024-02-10:
> why? WEMOS32 is just another ESP32 ...

Because I don't use ESP32 In my projects. I am not sure I can even find a ESP32 board.

SdFat by default uses a wrapper for the ESP driver that someone provided as a work around for a bug a number of years ago,

You could try editing SdFatConfig.h and changing this define so the wrapper is not used.

`#define SPI_DRIVER_SELECT 1  // default is zero.`

folkertvanheusden, 2024-02-14:
Got it to work with different pins.

closed 2024-02-14 by folkertvanheusden

folkertvanheusden, 2024-11-02:
On the WT32-ETH01 with sd-card on pin 12-15, I had to indeed use `#define SPI_DRIVER_SELECT 1` together with `SHARED_SPI`.