greiman/SdFat-beta #88: Rp2040SdioSetup.ino with "SdError 0X1E, 0X0"

lhm0, 2024-12-14:
I really appreciate the SDIO expansion for Rp2040. This is exactly what I need.

However, while both Rp2040SdioSetup.ino and bench.ino run on one of my raspberry pi picos flawlessly, I struggle with another. I am pretty sure it is NOT a wiring issue. 

Rp2040SdioSetup outputs:

begin() failed
Do not reformat the SD.
SdError: 0X1E,0X0

What does this error code mean? what could be the issue?

greiman, 2024-12-14:
This is a timeout while waiting for data from from the SD.

It is at one of three places in the RP2040  SDIO driver.

```
libraries\SdFat\src\SdCard\Rp2040Sdio\PioSdioCard.cpp (3 hits)
	Line 385:         sdError(SD_CARD_ERROR_READ_TIMEOUT);
	Line 465:           sdError(SD_CARD_ERROR_READ_TIMEOUT);
	Line 493:         sdError(SD_CARD_ERROR_READ_TIMEOUT);
```

>  I am pretty sure it is NOT a wiring issue.

It is could be a configuration problem. Do these match your wiring?
```
#define RP_CLK_GPIO <your CLK>
#define RP_CMD_GPIO <your CMD>
#define RP_DAT0_GPIO <Your DAT0>
```
Another possibility is that the wiring/socket distorts the signals since the timing is critical for SDIO.

I can't help much since I can't look at your problem with a scope or logic analyzer.

lhm0, 2024-12-14:
thank you for your reply.

Unfortunately, my scope is too slow for a meaningful analysis. I can see 30 MHz clock pulses, but the edges are flattened out very much by the bandwidth of the scope.

Your guess regarding a distortion is probably right. I have modified an existing pcb that used to use SPI for addressing the SD card. I changed the control lines for SDIO control and ended up with relatively long wires (10 cm, maybe) without proper return paths. Probably, there is lots of cross talk between the lines and other logic. 

With that in mind, it probably does not make much sense to dig deeper with a logic analyzer. It is likely, that the cause is not software, but hardware.

I will include SDIO control in the next pcb revision and see, if it works.